zhulf0804/PointPillars #2: 开启模型训练的时候出现了错误

hello-lx, 2022-06-15:
您好，我在处理预处理的时候并未发现异常，但是在准备训练的时候出现了维度不匹配的问题，不知道是哪里出现问题了，能否帮忙看看呢

PS E:\XSpace\DeepLearning\PointPillars> python train.py --data_root  D:\BaiduNetdiskDownload\object
==================== 0 ====================
  0%|                                                                                                                                                          | 0/619 [00:00<?, ?it/s]>
 e:\xspace\deeplearning\pointpillars\model\anchors.py(183)anchor_target()
-> assigned_gt_labels_weights = torch.zeros_like(cur_anchors[:, 0])
(Pdb) pos_flag
tensor([False, False, False,  ..., False, False, False], device='cuda:0')
(Pdb) pos_flag.shape
torch.Size([107136])
(Pdb) assigned_gt_inds.shape
torch.Size([107136])
(Pdb) gt_labels
tensor([-1,  2,  2,  2,  2,  2,  2,  2,  2,  2,  2,  2,  0,  0,  0,  0,  0,  0,
         0,  0,  0,  1,  1,  1,  1,  1,  1], device='cuda:0',
       dtype=torch.int32)
(Pdb) assigned_gt_inds[pos_flag] - 1
tensor([26, 26, 26, 25, 25, 24, 12, 24, 12, 24, 18, 18, 17, 17, 16, 16, 13, 13,
        22, 22, 15, 19, 19, 20, 20, 20, 20, 14], device='cuda:0')

zhulf0804, 2022-06-15:
出错信息是什么呢 ?

hello-lx, 2022-06-15:
维度不匹配

PS E:\XSpace\DeepLearning\PointPillars> python train.py
==================== 0 ====================
  0%|                                                                                                                                                          | 0/619 [00:10<?, ?it/s]
Traceback (most recent call last):
  File "train.py", line 215, in <module>
    main(args)
  File "train.py", line 86, in main
    batched_gt_labels=batched_labels)
  File "E:\XStudio\Miniconda3\lib\site-packages\torch\nn\modules\module.py", line 889, in _call_impl
    result = self.forward(*input, **kwargs)
  File "E:\XSpace\DeepLearning\PointPillars\model\pointpillars.py", line 410, in forward
    nclasses=self.nclasses)
  File "E:\XSpace\DeepLearning\PointPillars\model\anchors.py", line 178, in anchor_target
    assigned_gt_labels[pos_flag] = gt_labels[assigned_gt_inds[pos_flag] - 1]
RuntimeError: Index put requires the source and destination dtypes match, got Long for the destination and Int for the source.

zhulf0804, 2022-06-15:
这是数据类型的问题, 不是维度不匹配的问题。

试一下转变数据类型 (.long()) ?
`assigned_gt_labels[pos_flag] = gt_labels[assigned_gt_inds[pos_flag] - 1].long()`

hello-lx, 2022-06-15:
谢谢，已经解决了

closed 2022-06-15 by hello-lx