zhulf0804/PointPillars #5: ops文件夹无法compile

zhaoyewei, 2022-07-19:
如题，可能是因为包兼容问题，也可能是因为其他问题，ops里面的ops无法compile。
我尝试寻找了mmdet3d，但是把这部分代码提取出来。
请问楼主，有没有已经编译好ops的项目目录呢？

zhulf0804, 2022-07-19:
你好, 我编译好的你不一定能用。

你那边在编译时报什么错误啊 ?

zhaoyewei, 2022-07-19:
> 你好, 我编译好的你不一定能用。
> 
> 你那边在编译时报什么错误啊 ?

提示我的环境变量里的cuda路径找不到，但是这个路径确实是有的，其他信息也没tracking出来。

zhaoyewei, 2022-07-20:
> > 你好, 我编译好的你不一定能用。

[image]
当然，我的是cuda 10.2，并不是requirements当中的cuda11+torch1.8.1。可能是这个原因吧。

zhulf0804, 2022-07-20:
应该不是cuda版本的问题, 我在cuda10.1也测试过代码, 是没有问题的。

你上面报的错我之前好像也遇到过, 是环境配置中的路径问题。

你看你的两个路径(cuda-10.2和nvcc对应的路径)连到一块了, 把这个解决了应该就好了。

可以参考这个[issue](https://github.com/NVIDIA/apex/issues/368)里的解决方案: https://github.com/NVIDIA/apex/issues/368#issuecomment-507533209.

zhaoyewei, 2022-07-20:
> 应该不是cuda版本的问题, 我在cuda10.1也测试过代码, 是没有问题的。
> 
> 你上面报的错我之前好像也遇到过, 是环境配置中的路径问题。
> 
> 你看你的两个路径(cuda-10.2和nvcc对应的路径)连到一块了, 把这个解决了应该就好了。
> 
> 可以参考这个[issue](https://github.com/NVIDIA/apex/issues/368)里的解决方案: [NVIDIA/apex#368 (comment)](https://github.com/NVIDIA/apex/issues/368#issuecomment-507533209).

嗯，确实是这个问题，已经解决，感谢！

closed 2022-07-20 by zhulf0804