zhulf0804/PointPillars #7: 能否推理自己的点云图片或者视频

evil-master, 2022-08-01:
请问，是否能使用rosbag包中保存的bin文件进行推理？如果使用自己的数据集推理，需要将点云数据保持和kitti数据大小一致才能完成推理吗？

zhulf0804, 2022-08-01:
你好, 可以使用命令 `python test.py --ckpt pretrained/epoch_160.pth --pc_path your_file.bin` 试试。
需要注意的是 (1) 你的bin文件需要正确的读入: (n, 4)的数组; (2) 你的bin文件的主要数据的range应该在[0, -39.68, -3, 69.12, 39.68, 1]。否则的话, 可能需要修改一下代码了。

evil-master, 2022-08-02:
> 你好, 可以使用命令 `python test.py --ckpt pretrained/epoch_160.pth --pc_path your_file.bin` 试试。 需要注意的是 (1) 你的bin文件需要正确的读入: (n, 4)的数组; (2) 你的bin文件的主要数据的range应该在[0, -39.68, -3, 69.12, 39.68, 1]。否则的话, 可能需要修改一下代码了。

谢谢，我看readme没有说明依赖库，打开requirement文件一个个安装库吗？其中pointpillars.egg==info这个库是怎么安装的，此外我的cuda版本版本是10.0可以运行test.py文件吗？我使用rosbag2kitti的代码将rosbag包转成了bin文件，但是有什么办法看到bin文件的rangge在[0, -39.68, -3, 69.12, 39.68, 1]吗？

zhulf0804, 2022-08-02:
1. requirements.txt是我的训练/测试环境, 你的环境已经有的可以不用安装。
2. pointpillars.egg==info是通过README里的[Compile]产生的。
3. Cuda10.1/Cuda11.1我测试过没有问题, Cuda10.0你可以试试。
4. 读入.bin, 基于numpy进行数值分析。

lkyahpu, 2022-08-26:
[image]
你好，测试时出现上面的错误。

zhulf0804, 2022-08-26:
Hello @lkyahpu,

I can't reproduce this bug on my computer.

Could you print the parameter type contained in `nms_gpu()`, such as `boxes`, `keep`, `thresh` and `boxes.device.index`.
https://github.com/zhulf0804/PointPillars/blob/9c98d51742ff21209e1a587cf6c6f680f1cc5224/ops/iou3d_module.py#L63-L64

lkyahpu, 2022-08-29:
好的，之前是在CPU上测试的，现在加载到GPU上可以正常测试了。
[image]

zhulf0804, 2022-08-29:
@lkyahpu  好的

closed 2022-09-08 by zhulf0804

aifpgahzw, 2024-03-19:
> 你好, 可以使用命令 `python test.py --ckpt pretrained/epoch_160.pth --pc_path your_file.bin` 试试。 需要注意的是 (1) 你的bin文件需要正确的读入: (n, 4)的数组; (2) 你的bin文件的主要数据的range应该在[0, -39.68, -3, 69.12, 39.68, 1]。否则的话, 可能需要修改一下代码了。

你好，请问一下这个数据范围[0, -39.68, -3, 69.12, 39.68, 1]分别代表什么意思，意思是点云x的范围为(0, -39.68)，y范围( -3, 69.12),z范围(39.68, 1)吗

aifpgahzw, 2024-03-19:
> 你好, 可以使用命令 `python test.py --ckpt pretrained/epoch_160.pth --pc_path your_file.bin` 试试。 需要注意的是 (1) 你的bin文件需要正确的读入: (n, 4)的数组; (2) 你的bin文件的主要数据的range应该在[0, -39.68, -3, 69.12, 39.68, 1]。否则的话, 可能需要修改一下代码了。

您好，点云已经按照这个范围过滤，用自己的数据测试，能输出检测框，但是点云空白的，如果单独显示bin文件是可以正常显示的，测试自己的bin文件不知道是不是修改一下哪里的代码才可以让点云正常显示出来。
[image]

aifpgahzw, 2024-03-19:
> > 你好, 可以使用命令 `python test.py --ckpt pretrained/epoch_160.pth --pc_path your_file.bin` 试试。 需要注意的是 (1) 你的bin文件需要正确的读入: (n, 4)的数组; (2) 你的bin文件的主要数据的range应该在[0, -39.68, -3, 69.12, 39.68, 1]。否则的话, 可能需要修改一下代码了。
> 
> 您好，点云已经按照这个范围过滤，用自己的数据测试，能输出检测框，但是点云空白的，如果单独显示bin文件是可以正常显示的，测试自己的bin文件不知道是不是修改一下哪里的代码才可以让点云正常显示出来。 ![QQ截图20240319221617](https://private-user-images.githubusercontent.com/90996332/314092020-bdf38978-5466-4fe0-897a-caeb387caf1f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA4NTgyMDMsIm5iZiI6MTcxMDg1NzkwMywicGF0aCI6Ii85MDk5NjMzMi8zMTQwOTIwMjAtYmRmMzg5NzgtNTQ2Ni00ZmUwLTg5N2EtY2FlYjM4N2NhZjFmLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAzMTklMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMzE5VDE0MTgyM1omWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWJkMjNhNjlmMjI3NGNkYjdiZTRkOGEwM2Q4MDY4NzBhOWQ4ODBhMDg1MDU5ZGNhYjQzZTA4MzlhZWRjYTZmODEmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.UP9pshHgkSnSryRArsuKev8-tA9ZdF_87Mfal4j51zE)

@zhulf0804

coolzz27, 2024-05-25:
> 你好，请问一下这个数据范围[0, -39.68, -3, 69.12, 39.68, 1]分别代表什么意思，意思是点云x的范围为(0, -39.68)，y范围( -3, 69.12),z范围(39.68, 1)吗

x in (0, 69.12), y in (-39.68, 39.68), z in (-3, 1)