zhulf0804/PointPillars #8: Train.py中的mask存在问题

KyleYueye, 2022-08-04:
第170行开始：
```
pos_idx = (batched_bbox_labels >= 0) & (batched_bbox_labels < args.nclasses)
bbox_pred = bbox_pred[pos_idx]
```

报错信息：`The shape of the mask [320000] at index 0 does not match the shape of the indexed tensor [240000, 7] at index 0`

我的 `batched_bbox_labels` 大小是 `[320000]`，`bbox_preds` 是 `[240000, 7]`

zhulf0804, 2022-08-05:
你好, 你在训练的时候是把args.nclasses=3变成了args.nclasses=4吗 ?

因为看你上面的shape是 240000 * 4 = 960000 = 320000 * 3.

如果是这样的话, 得需要增加anchors生成的数量(每一类别都有自己的anchors), 现在默认的是针对3类的。

对应的这里的n_anchors和n_classes也需要改一下。
https://github.com/zhulf0804/PointPillars/blob/a0e8af7204168a1823d8679cf2379576b4795a27/model/pointpillars.py#L185-L186

有问题再交流。

KyleYueye, 2022-08-05:
好的谢谢， 我的 `n_classes` 已经是4了，我把 `n_anchors` 改成8之后就顺利运行了，感谢！

closed 2022-08-05 by KyleYueye

Felixqzx, 2025-01-18:
> 你好, 你在训练的时候是把args.nclasses=3变成了args.nclasses=4吗 ?
> 
> 因为看你上面的shape是 240000 * 4 = 960000 = 320000 * 3.
> 
> 如果是这样的话, 得需要增加anchors生成的数量(每一类别都有自己的anchors), 现在默认的是针对3类的。
> 
> 对应的这里的n_anchors和n_classes也需要改一下。
> 
> [PointPillars/model/pointpillars.py](https://github.com/zhulf0804/PointPillars/blob/a0e8af7204168a1823d8679cf2379576b4795a27/model/pointpillars.py#L185-L186)
> 
> Lines 185 to 186 in [a0e8af7](/zhulf0804/PointPillars/commit/a0e8af7204168a1823d8679cf2379576b4795a27)
> 
>  class Head(nn.Module): 
>      def __init__(self, in_channel, n_anchors=6, n_classes=3): 
> 有问题再交流。

为啥我修改了没用，我原始数据集是多类别的，我预处理筛了一部分只考虑4个类别，然后也是参照这个类别数修改，是数据预处理筛选类别后造成的影响吗？

Felixqzx, 2025-01-18:
> > 你好, 你在训练的时候是把args.nclasses=3变成了args.nclasses=4吗 ?
> > 因为看你上面的shape是 240000 * 4 = 960000 = 320000 * 3.
> > 如果是这样的话, 得需要增加anchors生成的数量(每一类别都有自己的anchors), 现在默认的是针对3类的。
> > 对应的这里的n_anchors和n_classes也需要改一下。
> > [PointPillars/model/pointpillars.py](https://github.com/zhulf0804/PointPillars/blob/a0e8af7204168a1823d8679cf2379576b4795a27/model/pointpillars.py#L185-L186)
> > Lines 185 to 186 in [a0e8af7](/zhulf0804/PointPillars/commit/a0e8af7204168a1823d8679cf2379576b4795a27)
> > class Head(nn.Module):
> > def **init**(self, in_channel, n_anchors=6, n_classes=3):
> > 有问题再交流。
> 
> 为啥我修改了没用，我原始数据集是多类别的，我预处理筛了一部分只考虑4个类别，然后也是参照这个类别数修改，是数据预处理筛选类别后造成的影响吗？以下是我的报错信息
Traceback (most recent call last):
  File "/root/autodl-tmp/PointPillars-main/train.py", line 222, in <module>
    main(args)
  File "/root/autodl-tmp/PointPillars-main/train.py", line 107, in main
    bbox_pred = bbox_pred[pos_idx]
                ~~~~~~~~~^^^^^^^^^
IndexError: The shape of the mask [1928448] at index 0 does not match the shape of the indexed tensor [2571264, 7] at index 0

Felixqzx, 2025-01-18:
> > > 你好, 你在训练的时候是把args.nclasses=3变成了args.nclasses=4吗 ?
> > > 因为看你上面的shape是 240000 * 4 = 960000 = 320000 * 3.
> > > 如果是这样的话, 得需要增加anchors生成的数量(每一类别都有自己的anchors), 现在默认的是针对3类的。
> > > 对应的这里的n_anchors和n_classes也需要改一下。
> > > [PointPillars/model/pointpillars.py](https://github.com/zhulf0804/PointPillars/blob/a0e8af7204168a1823d8679cf2379576b4795a27/model/pointpillars.py#L185-L186)
> > > Lines 185 to 186 in [a0e8af7](/zhulf0804/PointPillars/commit/a0e8af7204168a1823d8679cf2379576b4795a27)
> > > class Head(nn.Module):
> > > def **init**(self, in_channel, n_anchors=6, n_classes=3):
> > > 有问题再交流。
> > 
> > 
> >为啥我修改了没用，我原始数据集是多类别的，我预处理筛了一部分只考虑4个类别，然后也是参照这个类别数修改，是数据预处理筛选类别后造成的影响吗？以下是我的报错信息
> > Traceback (most recent call last):
> > File "/root/autodl-tmp/PointPillars-main/train.py", line 222, in 
> > main(args)
> > File "/root/autodl-tmp/PointPillars-main/train.py", line 107, in main
> > bbox_pred = bbox_pred[pos_idx]
> > ~~~~~~~~~^^^^^^^^^
> > IndexError: The shape of the mask [1928448] at index 0 does not match the shape of the indexed tensor [2571264, 7] at index 0

Felixqzx, 2025-01-18:
> 你好, 你在训练的时候是把args.nclasses=3变成了args.nclasses=4吗 ?
>
> 因为看你上面的shape是 240000 * 4 = 960000 = 320000 * 3.
>
> 如果是这样的话, 得需要增加anchors生成的数量(每一类别都有自己的anchors), 现在默认的是针对3类的。
>
> 对应的这里的n_anchors和n_classes也需要改一下。 [...]

> 为啥我修改了没用，我原始数据集是多类别的，我预处理筛了一部分只考虑4个类别，然后也是参照这个类别数修改，是数据预处理筛选类别后造成的影响吗？以下是我的报错信息
> Traceback (most recent call last):
> File "/root/autodl-tmp/PointPillars-main/train.py", line 222, in
> main(args)
> File [...]