zhulf0804/PointPillars #10: TypeError: expected string or bytes-like object, How to solve?

ianphoi, 2022-08-12:
No CUDA runtime is found, using CUDA_HOME='C:/Program Files/NVIDIA GPU Computing Toolkit/CUDA\v11.7'
running develop
C:\Users\ianph\anaconda3\envs\SimplePointPillar\lib\site-packages\setuptools\command\easy_install.py:144: EasyInstallDeprecationWarning: easy_install command is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
C:\Users\ianph\anaconda3\envs\SimplePointPillar\lib\site-packages\setuptools\command\install.py:34: SetuptoolsDeprecationWarning: setup.py install is deprecated. Use build and pip and other standards-based tools.
  warnings.warn(
running egg_info
writing pointpillars.egg-info\PKG-INFO
writing dependency_links to pointpillars.egg-info\dependency_links.txt
writing top-level names to pointpillars.egg-info\top_level.txt
C:\Users\ianph\anaconda3\envs\SimplePointPillar\lib\site-packages\torch\utils\cpp_extension.py:411: UserWarning: Attempted to use ninja as the BuildExtension backend but we could not find ninja.. Falling back to using the slow distutils backend.
  warnings.warn(msg.format('we could not find ninja.'))
reading manifest file 'pointpillars.egg-info\SOURCES.txt'
writing manifest file 'pointpillars.egg-info\SOURCES.txt'
running build_ext
C:\Users\ianph\anaconda3\envs\SimplePointPillar\lib\site-packages\torch\utils\cpp_extension.py:346: UserWarning: Error checking compiler version for cl: [WinError 2] The system cannot find the file specified
  warnings.warn(f'Error checking compiler version for {compiler}: {error}')
Traceback (most recent call last):
  File "setup.py", line 4, in <module>
    setup(
  File "C:\Users\ianph\anaconda3\envs\SimplePointPillar\lib\site-packages\setuptools\__init__.py", line 87, in setup
    return distutils.core.setup(**attrs)
  File "C:\Users\ianph\anaconda3\envs\SimplePointPillar\lib\site-packages\setuptools\_distutils\core.py", line 148, in setup
    return run_commands(dist)
  File "C:\Users\ianph\anaconda3\envs\SimplePointPillar\lib\site-packages\setuptools\_distutils\core.py", line 163, in run_commands
    dist.run_commands()
  File "C:\Users\ianph\anaconda3\envs\SimplePointPillar\lib\site-packages\setuptools\_distutils\dist.py", line 967, in run_commands
    self.run_command(cmd)
  File "C:\Users\ianph\anaconda3\envs\SimplePointPillar\lib\site-packages\setuptools\dist.py", line 1214, in run_command
    super().run_command(command)
  File "C:\Users\ianph\anaconda3\envs\SimplePointPillar\lib\site-packages\setuptools\_distutils\dist.py", line 986, in run_command
    cmd_obj.run()
  File "C:\Users\ianph\anaconda3\envs\SimplePointPillar\lib\site-packages\setuptools\command\develop.py", line 34, in run
    self.install_for_development()
  File "C:\Users\ianph\anaconda3\envs\SimplePointPillar\lib\site-packages\setuptools\command\develop.py", line 114, in install_for_development
    self.run_command('build_ext')
  File "C:\Users\ianph\anaconda3\envs\SimplePointPillar\lib\site-packages\setuptools\_distutils\cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "C:\Users\ianph\anaconda3\envs\SimplePointPillar\lib\site-packages\setuptools\dist.py", line 1214, in run_command
    super().run_command(command)
  File "C:\Users\ianph\anaconda3\envs\SimplePointPillar\lib\site-packages\setuptools\_distutils\dist.py", line 986, in run_command
    cmd_obj.run()
  File "C:\Users\ianph\anaconda3\envs\SimplePointPillar\lib\site-packages\setuptools\command\build_ext.py", line 79, in run
    _build_ext.run(self)
  File "C:\Users\ianph\anaconda3\envs\SimplePointPillar\lib\site-packages\setuptools\_distutils\command\build_ext.py", line 339, in run
    self.build_extensions()
  File "C:\Users\ianph\anaconda3\envs\SimplePointPillar\lib\site-packages\torch\utils\cpp_extension.py", line 434, in build_extensions
    self._check_cuda_version(compiler_name, compiler_version)
  File "C:\Users\ianph\anaconda3\envs\SimplePointPillar\lib\site-packages\torch\utils\cpp_extension.py", line 808, in _check_cuda_version
    torch_cuda_version = packaging.version.parse(torch.version.cuda)
  File "C:\Users\ianph\anaconda3\envs\SimplePointPillar\lib\site-packages\pkg_resources\_vendor\packaging\version.py", line 49, in parse
    return Version(version)
  File "C:\Users\ianph\anaconda3\envs\SimplePointPillar\lib\site-packages\pkg_resources\_vendor\packaging\version.py", line 264, in __init__
    match = self._regex.search(version)
TypeError: expected string or bytes-like object

zhulf0804, 2022-08-14:
Hello, I also find the same error in https://github.com/Spandan-Madan/Pytorch_fine_tuning_Tutorial/issues/10#issuecomment-499406167.

Try reducing the number of workers in DataLoader by setting `--num_workers 1` ?

ianphoi, 2022-08-15:
i change the num worker = 0, and the problem is solved, but now a new problem arise

ianphoi, 2022-08-15:
(SimplePointPillar) C:\Users\ianph\SimplePointPillar>python train.py --data_root your_path_to_kitti
==================== 0 ====================
  0%|                                                                                                                                                                                 | 0/619 [00:00<?, ?it/s]
Traceback (most recent call last):
  File "train.py", line 215, in <module>
    main(args)
  File "train.py", line 68, in main
    for i, data_dict in enumerate(tqdm(train_dataloader)):
  File "C:\Users\ianph\anaconda3\envs\SimplePointPillar\lib\site-packages\tqdm\std.py", line 1180, in __iter__
    for obj in iterable:
  File "C:\Users\ianph\anaconda3\envs\SimplePointPillar\lib\site-packages\torch\utils\data\dataloader.py", line 517, in __next__
    data = self._next_data()
  File "C:\Users\ianph\anaconda3\envs\SimplePointPillar\lib\site-packages\torch\utils\data\dataloader.py", line 557, in _next_data
    data = self._dataset_fetcher.fetch(index)  # may raise StopIteration
  File "C:\Users\ianph\anaconda3\envs\SimplePointPillar\lib\site-packages\torch\utils\data\_utils\fetch.py", line 44, in fetch
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "C:\Users\ianph\anaconda3\envs\SimplePointPillar\lib\site-packages\torch\utils\data\_utils\fetch.py", line 44, in <listcomp>
    data = [self.dataset[idx] for idx in possibly_batched_index]
  File "C:\Users\ianph\SimplePointPillar\dataset\kitti.py", line 104, in __getitem__
    pts = read_points(pts_path)
  File "C:\Users\ianph\SimplePointPillar\utils\io.py", line 22, in read_points
    return np.fromfile(file_path, dtype=np.float32).reshape(-1, dim)
FileNotFoundError: [Errno 2] No such file or directory: 'your_path_to_kitti\\your_path_to_kitti\\training\\velodyne_reduced\\004037.bin'

ianphoi, 2022-08-15:
the file is located at your_path_to_kitti\training\velodyne_reduced\004037.bin'
and not //FileNotFoundError: [Errno 2] No such file or directory: 'your_path_to_kitti\your_path_to_kitti\training\velodyne_reduced\004037.bin'//

ianphoi, 2022-08-15:
[image]

zhulf0804, 2022-08-15:
Maybe there are some path problems when you pre-process KITTI dataset.

Now, you can try `python train.py --data_root ./` if your KITTI dataset is under `your_path_to_kitti` or `./your_path_to_kitti`, as you mentioned above.

ianphoi, 2022-08-15:
(SimplePointPillar) C:\Users\ianph\SimplePointPillar>python train.py --data_root ./
Traceback (most recent call last):
  File "train.py", line 215, in <module>
    main(args)
  File "train.py", line 25, in main
    train_dataset = Kitti(data_root=args.data_root,
  File "C:\Users\ianph\SimplePointPillar\dataset\kitti.py", line 49, in __init__
    self.data_infos = read_pickle(os.path.join(data_root, f'kitti_infos_{split}.pkl'))
  File "C:\Users\ianph\SimplePointPillar\utils\io.py", line 8, in read_pickle
    with open(file_path, 'rb') as f:
FileNotFoundError: [Errno 2] No such file or directory: './kitti_infos_train.pkl'

(SimplePointPillar) C:\Users\ianph\SimplePointPillar>

ianphoi, 2022-08-15:
i tried to add ./ into the pkl file but the it does not allow special character

ianphoi, 2022-08-15:
nvm, i just rename my file as data and change my --data root data and solve this issue

ianphoi, 2022-08-15:
[image]

ianphoi, 2022-08-15:
Do you think i need to redownload the file since FileNotFoundError: Caught FileNotFoundError in DataLoader worker process 0. which may cause FileNotFoundError: [Errno 2] No such file or directory: 'data\\data\\training\\velodyne_reduced\\004037.bin
[image]
'

ianphoi, 2022-08-15:
Still have the same error
FileNotFoundError: [Errno 2] No such file or directory: 'data\\data\\training\\velodyne_reduced\\004037.bin'

ianphoi, 2022-08-16:
I found out that in \utils\io.py, in line 23, the file_path is data\data\training\velodyne_reduced\004037.bin

zhulf0804, 2022-08-16:
> I found out that in \utils\io.py, in line 23, the file_path is data\data\training\velodyne_reduced\004037.bin

It's strange about  your repetitive data path `data\data` and `your_path_to_kitti\your_path_to_kitti`. 

Did you pre-process KITTI by `python pre_process_kitti.py --data_root your_path_to_kitti`, where `your_path_to_kitti` is the absolute path to KITTI. 

`your_path_to_kitti` is the style of `/mnt/ssd1/lifa_rdata/det/kitti`, and the contents included in `/mnt/ssd1/lifa_rdata/det/kitti` can be seen [Datasets](https://github.com/zhulf0804/PointPillars#datasets).

ianphoi, 2022-08-16:
[image]
I think so, I run the code on anaconda terminal
activate SimplePointPillar
cd SimplePointPillar
cd ops
python setup.py develop
cd..
python pre_process_kitti.py --data_root data
python train.py --data_root data
python evaluate.py --ckpt pretrained/epoch_160.pth --data_root data

Hongsehyun, 2022-08-16:
now I have same issue... about  data path like `FileNotFoundError: [Errno 2] No such file or directory: 'kitti\\kitti\\training\\velodyne_reduced\\004037.bin'`

when i ran this code : `python train.py --data_root kitti`

and it's strange... if i ran this code : `python train.py --data_root kitti/`

this error was presented

`FileNotFoundError: [Errno 2] No such file or directory: 'kitti/kitti\\training\\velodyne_reduced\\004037.bin'`

I composed the Kitti data path just like this 
[image]

Did anyone solve it? thanks in advance...

zhulf0804, 2022-08-16:
> now I have same issue... about data path like `FileNotFoundError: [Errno 2] No such file or directory: 'kitti\\kitti\\training\\velodyne_reduced\\004037.bin'`
>
> when i ran this code : `python [...]

Hello, the path problem may be related to OS. I'll check the code using Windows later.

Best.

ianphoi, 2022-08-17:
> Hello, the path problem may be related to OS. I'll check the code using Windows later.
>
> Best.

how is the progress??

zhulf0804, 2022-08-18:
Hi @ianphoi,

Sorry for the late check for I'm busy these days.

I'll check it as soon as possible (maybe tomorrow).

zhulf0804, 2022-08-18:
Hi @ianphoi,

I find the reason for the above path problem. The path separator `/` and `\` between Linux and Windows are different. The code has only been tested in Linux before. 

So try the following 2 steps:
1. Replace **all** `'/'` with `os.path.sep` in **pre_process_kitti.py**, for example, `'/'.join(lidar_path.split('/')[-3:])` -> `os.path.sep.join(lidar_path.split(os.path.sep)[-3:])`
2. Re-process KITTI dataset with `python pre_process_kitti.py --data_root your_path_to_kitti`

(Windows is not available to me now. If there are any problems about the above steps, please let me know.)

Best.

ianphoi, 2022-08-18:
Yup, thank you the problem is solved

closed 2022-08-18 by ianphoi