zhulf0804/PointPillars #13: RuntimeError: CUDA out of memory

ianphoi, 2022-08-18:
[image]
so i reduce my batch_size = 3 and num_worker = 1
then i gave me this error
[image]

zhulf0804, 2022-08-18:
Hello @ianphoi, 

please refer to this issue https://github.com/zhulf0804/PointPillars/issues/2#issuecomment-1155941738.

ianphoi, 2022-08-18:
thank you, solved it

ianphoi, 2022-08-19:
How to run the train.py in a GPU?
Since when I was running the train.py with batch size of 1, it take forever and I notice that it runs on my cpu and not gpu

zhulf0804, 2022-08-19:
It runs on GPU by default.

ianphoi, 2022-08-20:
how many epoch does I need to train because the program have been running for 20 hours and it is currently at epoch 63. Should i change my num worker and batch size. Currently my num worker =1, batch size = 2

ianphoi, 2022-08-20:
is it possible to pause the train.py which window can do by pressing control + pause and the change the num worker and batch size then continue to resume the program?

zhulf0804, 2022-08-20:
Hi @ianphoi,

1. 160 epochs. It takes about 20 hours to train the model (160 epochs) on one RTX 3090 GPU card with the default parameters.
2. Not yet. I'll update the code by supporting model resuming in the future.

Best.

ianphoi, 2022-08-20:
> Hi @ianphoi,
> 
> 1. 160 epochs. It takes about 20 hours to train the model (160 epochs) on one RTX 3090 GPU card with the default parameters.
> 2. Not yet. I'll update the code by supporting model resuming in the future.
> 
> Best.

alright thank you. You mention that  it uses GPU right. But when I was running the train.py, my task manager indicates zero usage of GPU.

closed 2022-09-04 by ianphoi