zhulf0804/PointPillars #16: 360 Degree detection issue

CrazyMage24, 2022-09-08:
Hi,

First of all, thanks for the work, the program works really well for me.
I am using the pretrained network on my own LiDAR datasets.
I noticed that the point cloud visualization and detection only works for the front half.

I changed the test.py to

<html><body>

<sub> def point_range_filter(pts, point_range=[-69.12, -39.68, -3, 69.12, 39.68, 1]): </sub> |  
-- | --

</body>
</html>

and now it visualizes the whole point cloud, but the detection still only applies to the front half.
Can you help with that?

Current progress:
[image]

Thanks in advance!

zhulf0804, 2022-09-08:
Hi @CrazyMage24,

Try to update the `point_cloud_range` parameter in the following 3 code blocks,

1. https://github.com/zhulf0804/PointPillars/blob/9c98d51742ff21209e1a587cf6c6f680f1cc5224/model/pointpillars.py#L220-L226
2. https://github.com/zhulf0804/PointPillars/blob/9c98d51742ff21209e1a587cf6c6f680f1cc5224/test.py#L15
3. https://github.com/zhulf0804/PointPillars/blob/9c98d51742ff21209e1a587cf6c6f680f1cc5224/test.py#L38

Please let me know if there are any problems, thanks.

zhulf0804, 2022-09-09:
Hello @CrazyMage24,

Could you please send the tested Lidar data to my mailbox (zhulf0804@gmail.com) if possible?

I'll debug it in the next few days.

Thanks.

CrazyMage24, 2022-09-09:
Hi,

I have a Drive folder with the bin files
https://drive.google.com/drive/folders/1HAAgm9dSs5PbshG-mb4CV-bG7DgoOwyK?usp=sharing

Thank you!

zhulf0804, 2022-09-11:
Hi @CrazyMage24,

Maybe there are **4** code blocks to be updated for point cloud range.

**In addition to** the above 3 code blocks, **the 4th** code block needed to be updated is
https://github.com/zhulf0804/PointPillars/blob/9c98d51742ff21209e1a587cf6c6f680f1cc5224/model/pointpillars.py#L245-L248

Please let me know if there are any problems, thank you.

closed 2022-11-13 by zhulf0804