zhulf0804/PointPillars #17: python setup.py develop compile error

hoangduyloc, 2022-09-13:
Hi, Thanks for your great work.
I have problem when compile  the "python setup develop" command.
I used conda env, GPU 3080Ti. Do you have exp to handle that GPU, also Might I ask what is your GPU, and envs?
Thank you.

Some part of the errors:
"  File "/home/abc/anaconda3/envs/PP/lib/python3.8/site-packages/setuptools/command/build_ext.py", line 202, in build_extension
    _build_ext.build_extension(self, ext)
  File "/home/abc/anaconda3/envs/PP/lib/python3.8/site-packages/setuptools/_distutils/command/build_ext.py", line 547, in build_extension
    objects = self.compiler.compile(
  File "/home/abc/anaconda3/envs/PP/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 529, in unix_wrap_ninja_compile
    _write_ninja_file_and_compile_objects(
  File "/home/abc/anaconda3/envs/PP/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 1354, in _write_ninja_file_and_compile_objects
    _run_ninja_build(
  File "/home/abc/anaconda3/envs/PP/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 1683, in _run_ninja_build
    raise RuntimeError(message) from e
RuntimeError: Error compiling objects for extension"

zhulf0804, 2022-09-13:
Hi @hoangduyloc,

I tested on RTX3090 and  TITAN X. GTX1080 was also tested (https://github.com/zhulf0804/PointPillars/issues/4#issuecomment-1217373258) by [cadwallader113](https://github.com/cadwallader113). 

Which PyTorch version do you use? I find the above error may be related PyTorch version.

Refer to https://github.com/facebookresearch/maskrcnn-benchmark/issues/1236 and https://github.com/CharlesShang/DCNv2/issues/86.

Best.

hoangduyloc, 2022-09-13:
Thanks for your fast answer,
I installed torch locally via

hoangduyloc, 2022-09-13:
Hello, thanks for your fast answer, I install torch locally via https://pytorch.org/get-started/locally/ - "conda install pytorch torchvision torchaudio cudatoolkit=11.3 -c pytorch", and that torch version is 1.12.1
Also, I'm using Ubuntu 22; it is difficult to find documents for installing Cuda, CuDnn in custom version.

Additionally, might I ask which Ubuntu version and Env like Conda are you using? Thank you

zhulf0804, 2022-09-13:
Hello @hoangduyloc, I'm using Ubuntu 18.04.4 LTS with Conda (PyTorch1.8.1+cu111, Python3.8) and RTX3090 cards.

hoangduyloc, 2022-09-13:
> Hello @hoangduyloc, I'm using Ubuntu 18.04.4 LTS with Conda (PyTorch1.8.1+cu111, Python3.8) and RTX3090 cards.

Thank you. I will try abit with Ubuntu 22.04; Otherwise I need to downgrade to Ubuntu 18.04 like yours :D. I really appreciate your work and your help, thanks.

zhulf0804, 2022-09-13:
Ok, I think you can.

Best regards.

closed 2022-09-18 by hoangduyloc

ouguangjun, 2024-05-31:
hi