zhulf0804/PointPillars #18: Cannot Display the grouptruth --calib_path

ianphoi, 2022-09-20:
python test.py --ckpt pretrained/epoch_160.pth --pc_path your_pc_path --calib_path your_calib_path|
it produced this error
[Open3D WARNING] [ViewControl] ConvertFromPinholeCameraParameters() failed because window height and width do not match.
how to solve it so I visualize the groundtruth?

zhulf0804, 2022-09-22:
Hi @ianphoi ,

Try to delete the following 3 lines in `utils/vis_o3d.py` and then re-run `python test.py --ckpt pretrained/epoch_160.pth --pc_path your_pc_path --calib_path your_calib_path --gt_path your_gt_path`.

```
ctr = vis.get_view_control()
param = o3d.io.read_pinhole_camera_parameters(os.path.join(PAR, 'viewpoint.json'))
ctr.convert_from_pinhole_camera_parameters(param)
```

Best.

ianphoi, 2022-09-22:
Ok, thank you. I found out that it is not the code problem but I did not include --gt_path your_gt_path as the # 2. infer and visualize point cloud detection and gound truth did not mention. Thank you