zhulf0804/PointPillars #19: ModuleNotFoundError: No module named 'ops.voxel_op'

sparkssjj, 2022-09-22:
你好，我遇到如下两个问题，这个是什么原因：
1.执行evaluate.py脚本时报错：
PointPillars/ops/voxel_module.py", line 5, in <module>
    from .voxel_op import hard_voxelize
ModuleNotFoundError: No module named 'ops.voxel_op'

2.执行python setup.py develop时，卡住不动，不报错也没有任何提示。

zhulf0804, 2022-09-22:
你好,

第一个问题是由于没有编译导致的, 通过执行`python setup.py develop`解决;

第二个问题, 暂时没有遇到过; 卡住了多久 ?

sparkssjj, 2022-09-22:
卡了几个小时，猜测可能是torch版本，cuda版本导致，请问你的cuda是什么版本

zhulf0804, 2022-09-22:
10.1和11.1。

JachinCai, 2022-09-25:
您好，请问有没有碰到这个我这个问题，执行python setup.py develop生成两个动态库之后pycharm还是识别不了
Unresolved reference 'iou3d_op' 
Unresolved reference 'voxel_op'

JachinCai, 2022-09-25:
iou3d_op.cpython-38-x86_64-linux-gnu.so
voxel_op.cpython-38-x86_64-linux-gnu.so

daaiwusheng, 2022-10-07:
@JachinCai 同样的问题, pycharm识别不了,请问您解决了吗?

zhulf0804, 2022-10-08:
在命令行加入相应的路径并运行试试 ?

daaiwusheng, 2022-10-08:
> 

命令行运行没问题,就是pycharm 索引不到这两个库, 尝试了各种办法,还是没有解决.

daaiwusheng, 2022-10-08:
> 在命令行加入相应的路径并运行试试 ?

命令行运行没问题,就是pycharm 索引不到这两个库, 尝试了各种办法,还是没有解决.

AlbertFeng-0405, 2022-12-13:
Try to change .voxel_op tp voxel_op. (Remove ".")

yanlibaba, 2023-03-04:
> 你好，我遇到如下两个问题，这个是什么原因： 1.执行evaluate.py脚本时报错： PointPillars/ops/voxel_module.py", line 5, in from .voxel_op import hard_voxelize ModuleNotFoundError: No module named 'ops.voxel_op'
> 
> 2.执行python setup.py develop时，卡住不动，不报错也没有任何提示。

请问您的第2个问题是怎么解决的

stu-zhangxl, 2023-03-21:
Excuse me, teacher. I operate in an environment without a GPU. How can I conduct testing? Currently, I want to only run the test file, but some feedback information is missing voxel_op & iou3d_op. Can I test the data without compiling the gpu environment?

szyilmaz, 2023-04-04:
If you are trying to make this code work in windows you must install "Desktop Development With C++" 
[image]