zhulf0804/PointPillars #21: TypeError: tuple indices must be integers or slices, not str

azooz95, 2022-09-26:
Thank you for your remarkable efforts.

I am running the test file for inference, however i am getting typeError as following: 
**Traceback (most recent call last):
  File "/home/azooz/Downloads/PointPillars/test.py", line 138, in <module>
    main(args)
  File "/home/azooz/Downloads/PointPillars/test.py", line 83, in main
    result_filter = keep_bbox_from_lidar_range(result_filter, pcd_limit_range)
  File "/home/azooz/Downloads/PointPillars/utils/process.py", line 592, in keep_bbox_from_lidar_range
    lidar_bboxes =  result['lidar_bboxes']
TypeError: tuple indices must be integers or slices, not str**

Thank you in advance.

RunTian1, 2022-09-30:
> Thank you for your remarkable efforts.
> 
> I am running the test file for inference, however i am getting typeError as following: **Traceback (most recent call last): File "/home/azooz/Downloads/PointPillars/test.py", line 138, in main(args) File "/home/azooz/Downloads/PointPillars/test.py", line 83, in main result_filter = keep_bbox_from_lidar_range(result_filter, pcd_limit_range) File "/home/azooz/Downloads/PointPillars/utils/process.py", line 592, in keep_bbox_from_lidar_range lidar_bboxes = result['lidar_bboxes'] TypeError: tuple indices must be integers or slices, not str**
> 
> Thank you in advance.

I have encountered this problem before. The main reason is that the pillars returned by PillarsLayer had a value of 0, which was later used as the denominator when computing the center point in PillarEncoder, resulting in nan results. Find the part of pillars with a value of 0 to solve it

zhulf0804, 2022-10-01:
Hello @azooz95,

Did you modify the original code? I find line 592 in file "PointPillars/utils/process.py" is not `lidar_bboxes = result['lidar_bboxes']`.

Could you show me the complete code here ? Thanks.

ye12121, 2022-11-16:
> Hello @azooz95,
> 
> Did you modify the original code? I find line 592 in file "PointPillars/utils/process.py" is not `lidar_bboxes = result['lidar_bboxes']`.
> 
> Could you show me the complete code here ? Thanks.

我也遇到这个问题，报错也是在这行代码 lidar_bboxes = result['lidar_bboxes']，原因是result不是dict，返回的是一个空的tuple，应该是模型推理过程中检测物体，对应的代码应该是 pointpillars.py中337行 return [], [], []。评估函数里没有对异常值做处理所以报错了。

quliang93, 2023-05-07:
> Hello @azooz95,
> 
> Did you modify the original code? I find line 592 in file "PointPillars/utils/process.py" is not `lidar_bboxes = result['lidar_bboxes']`.
> 
> Could you show me the complete code here ? Thanks.
Thank you for your reply ! 
Great Job !
This problem didn't happen in official KITTI dataset. But after I tested it in my custom dataset. I also met this error.
I did some test in a special scene. If the scene didn't contain any objects like Car, Pedestrian or Cyclist, the model output ([], [], []). And I got the same error like @azooz95 mentioned.

Can you give me some advice? Thank you very much

quliang93, 2023-05-07:
> > Hello @azooz95,
> > Did you modify the original code? I find line 592 in file "PointPillars/utils/process.py" is not `lidar_bboxes = result['lidar_bboxes']`.
> > Could you show me the complete code here ? Thanks.
> 
> 我也遇到这个问题，报错也是在这行代码 lidar_bboxes = result['lidar_bboxes']，原因是result不是dict，返回的是一个空的tuple。应该是模型推理过程中没有检测到物体。对应的代码应该是 pointpillars.py中337行 return [], [], []。评估函数里没有对异常值做处理所以报错了。

确实如此，请问您解决这个问题了吗？

quliang93, 2023-05-07:
> 

确实是这样。请问您解决这个问题了吗？

closed 2024-04-06 by zhulf0804