zhulf0804/PointPillars #22: "IndexError: list index out of range"

hs4449889, 2022-09-27:
Error when running setup.py in the following environment
> Conda(pytorch1.8.1,python3.8,cuda11.1) 

The error message is as follows
I don't know the cause of the error, please help me.
> No CUDA runtime is found, using CUDA_HOME='/apps/cuda/11.3.1'
running develop
running egg_info
writing pointpillars.egg-info/PKG-INFO
writing dependency_links to pointpillars.egg-info/dependency_links.txt
writing top-level names to pointpillars.egg-info/top_level.txt
/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/torch/utils/cpp_extension.py:411: UserWarning: Attempted to use ninja as the BuildExtension backend but we could not find ninja.. Falling back to using the slow distutils backend.
  warnings.warn(msg.format('we could not find ninja.'))
reading manifest file 'pointpillars.egg-info/SOURCES.txt'
writing manifest file 'pointpillars.egg-info/SOURCES.txt'
running build_ext
building 'voxel_op' extension
gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -DWITH_CUDA -I/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/torch/include -I/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/torch/include/TH -I/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/torch/include/THC -I/apps/cuda/11.3.1/include -I/apps/cudnn/8.4.1/cuda11.3/include -I/home/ace14496ms/pypy38_pp/include -I/apps/centos7/python/3.8.13_gcc11.2.0/include/python3.8 -c voxelization/voxelization.cpp -o build/temp.linux-x86_64-3.8/voxelization/voxelization.o -DTORCH_API_INCLUDE_EXTENSION_H -DPYBIND11_COMPILER_TYPE="_gcc" -DPYBIND11_STDLIB="_libstdcpp" -DPYBIND11_BUILD_ABI="_cxxabi1011" -DTORCH_EXTENSION_NAME=voxel_op -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++14
gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -DWITH_CUDA -I/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/torch/include -I/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/torch/include/TH -I/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/torch/include/THC -I/apps/cuda/11.3.1/include -I/apps/cudnn/8.4.1/cuda11.3/include -I/home/ace14496ms/pypy38_pp/include -I/apps/centos7/python/3.8.13_gcc11.2.0/include/python3.8 -c voxelization/voxelization_cpu.cpp -o build/temp.linux-x86_64-3.8/voxelization/voxelization_cpu.o -DTORCH_API_INCLUDE_EXTENSION_H -DPYBIND11_COMPILER_TYPE="_gcc" -DPYBIND11_STDLIB="_libstdcpp" -DPYBIND11_BUILD_ABI="_cxxabi1011" -DTORCH_EXTENSION_NAME=voxel_op -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++14
Traceback (most recent call last):
  File "setup.py", line 4, in <module>
    setup(
  File "/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/setuptools/__init__.py", line 153, in setup
    return distutils.core.setup(**attrs)
  File "/apps/centos7/python/3.8.13_gcc11.2.0/lib/python3.8/distutils/core.py", line 148, in setup
    dist.run_commands()
  File "/apps/centos7/python/3.8.13_gcc11.2.0/lib/python3.8/distutils/dist.py", line 966, in run_commands
    self.run_command(cmd)
  File "/apps/centos7/python/3.8.13_gcc11.2.0/lib/python3.8/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/setuptools/command/develop.py", line 34, in run
    self.install_for_development()
  File "/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/setuptools/command/develop.py", line 136, in install_for_development
    self.run_command('build_ext')
  File "/apps/centos7/python/3.8.13_gcc11.2.0/lib/python3.8/distutils/cmd.py", line 313, in run_command
    self.distribution.run_command(command)
  File "/apps/centos7/python/3.8.13_gcc11.2.0/lib/python3.8/distutils/dist.py", line 985, in run_command
    cmd_obj.run()
  File "/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/setuptools/command/build_ext.py", line 79, in run
    _build_ext.run(self)
  File "/apps/centos7/python/3.8.13_gcc11.2.0/lib/python3.8/distutils/command/build_ext.py", line 340, in run
    self.build_extensions()
  File "/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 765, in build_extensions
    build_ext.build_extensions(self)
  File "/apps/centos7/python/3.8.13_gcc11.2.0/lib/python3.8/distutils/command/build_ext.py", line 449, in build_extensions
    self._build_extensions_serial()
  File "/apps/centos7/python/3.8.13_gcc11.2.0/lib/python3.8/distutils/command/build_ext.py", line 474, in _build_extensions_serial
    self.build_extension(ext)
  File "/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/setuptools/command/build_ext.py", line 196, in build_extension
    _build_ext.build_extension(self, ext)
  File "/apps/centos7/python/3.8.13_gcc11.2.0/lib/python3.8/distutils/command/build_ext.py", line 528, in build_extension
    objects = self.compiler.compile(sources,
  File "/apps/centos7/python/3.8.13_gcc11.2.0/lib/python3.8/distutils/ccompiler.py", line 574, in compile
    self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
  File "/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 513, in unix_wrap_single_compile
    cflags = unix_cuda_flags(cflags)
  File "/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 480, in unix_cuda_flags
    cflags + _get_cuda_arch_flags(cflags))
  File "/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 1694, in _get_cuda_arch_flags
    arch_list[-1] += '+PTX'
IndexError: list index out of range

zhulf0804, 2022-10-01:
Hello, did you install nvidia driver ?

yanlibaba, 2023-03-04:
> 

我也遇到这个问题了，查了下nvidia驱动是有的，怎么解决呢，希望您回复一下，非常感谢

yanlibaba, 2023-03-04:
> Error when running setup.py in the following environment
> 
> > Conda(pytorch1.8.1,python3.8,cuda11.1)
> 
> The error message is as follows I don't know the cause of the error, please help me.
> 
> > No CUDA runtime is found, using CUDA_HOME='/apps/cuda/11.3.1'
> > running develop
> > running egg_info
> > writing pointpillars.egg-info/PKG-INFO
> > writing dependency_links to pointpillars.egg-info/dependency_links.txt
> > writing top-level names to pointpillars.egg-info/top_level.txt
> > /home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/torch/utils/cpp_extension.py:411: UserWarning: Attempted to use ninja as the BuildExtension backend but we could not find ninja.. Falling back to using the slow distutils backend.
> > warnings.warn(msg.format('we could not find ninja.'))
> > reading manifest file 'pointpillars.egg-info/SOURCES.txt'
> > writing manifest file 'pointpillars.egg-info/SOURCES.txt'
> > running build_ext
> > building 'voxel_op' extension
> > gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -DWITH_CUDA -I/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/torch/include -I/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/torch/include/TH -I/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/torch/include/THC -I/apps/cuda/11.3.1/include -I/apps/cudnn/8.4.1/cuda11.3/include -I/home/ace14496ms/pypy38_pp/include -I/apps/centos7/python/3.8.13_gcc11.2.0/include/python3.8 -c voxelization/voxelization.cpp -o build/temp.linux-x86_64-3.8/voxelization/voxelization.o -DTORCH_API_INCLUDE_EXTENSION_H -DPYBIND11_COMPILER_TYPE="_gcc" -DPYBIND11_STDLIB="_libstdcpp" -DPYBIND11_BUILD_ABI="_cxxabi1011" -DTORCH_EXTENSION_NAME=voxel_op -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++14
> > gcc -pthread -Wno-unused-result -Wsign-compare -DNDEBUG -g -fwrapv -O3 -Wall -fPIC -DWITH_CUDA -I/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/torch/include -I/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/torch/include/torch/csrc/api/include -I/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/torch/include/TH -I/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/torch/include/THC -I/apps/cuda/11.3.1/include -I/apps/cudnn/8.4.1/cuda11.3/include -I/home/ace14496ms/pypy38_pp/include -I/apps/centos7/python/3.8.13_gcc11.2.0/include/python3.8 -c voxelization/voxelization_cpu.cpp -o build/temp.linux-x86_64-3.8/voxelization/voxelization_cpu.o -DTORCH_API_INCLUDE_EXTENSION_H -DPYBIND11_COMPILER_TYPE="_gcc" -DPYBIND11_STDLIB="_libstdcpp" -DPYBIND11_BUILD_ABI="_cxxabi1011" -DTORCH_EXTENSION_NAME=voxel_op -D_GLIBCXX_USE_CXX11_ABI=0 -std=c++14
> > Traceback (most recent call last):
> > File "setup.py", line 4, in 
> > setup(
> > File "/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/setuptools/**init**.py", line 153, in setup
> > return distutils.core.setup(**attrs)
> > File "/apps/centos7/python/3.8.13_gcc11.2.0/lib/python3.8/distutils/core.py", line 148, in setup
> > dist.run_commands()
> > File "/apps/centos7/python/3.8.13_gcc11.2.0/lib/python3.8/distutils/dist.py", line 966, in run_commands
> > self.run_command(cmd)
> > File "/apps/centos7/python/3.8.13_gcc11.2.0/lib/python3.8/distutils/dist.py", line 985, in run_command
> > cmd_obj.run()
> > File "/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/setuptools/command/develop.py", line 34, in run
> > self.install_for_development()
> > File "/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/setuptools/command/develop.py", line 136, in install_for_development
> > self.run_command('build_ext')
> > File "/apps/centos7/python/3.8.13_gcc11.2.0/lib/python3.8/distutils/cmd.py", line 313, in run_command
> > self.distribution.run_command(command)
> > File "/apps/centos7/python/3.8.13_gcc11.2.0/lib/python3.8/distutils/dist.py", line 985, in run_command
> > cmd_obj.run()
> > File "/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/setuptools/command/build_ext.py", line 79, in run
> > _build_ext.run(self)
> > File "/apps/centos7/python/3.8.13_gcc11.2.0/lib/python3.8/distutils/command/build_ext.py", line 340, in run
> > self.build_extensions()
> > File "/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 765, in build_extensions
> > build_ext.build_extensions(self)
> > File "/apps/centos7/python/3.8.13_gcc11.2.0/lib/python3.8/distutils/command/build_ext.py", line 449, in build_extensions
> > self._build_extensions_serial()
> > File "/apps/centos7/python/3.8.13_gcc11.2.0/lib/python3.8/distutils/command/build_ext.py", line 474, in _build_extensions_serial
> > self.build_extension(ext)
> > File "/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/setuptools/command/build_ext.py", line 196, in build_extension
> > _build_ext.build_extension(self, ext)
> > File "/apps/centos7/python/3.8.13_gcc11.2.0/lib/python3.8/distutils/command/build_ext.py", line 528, in build_extension
> > objects = self.compiler.compile(sources,
> > File "/apps/centos7/python/3.8.13_gcc11.2.0/lib/python3.8/distutils/ccompiler.py", line 574, in compile
> > self._compile(obj, src, ext, cc_args, extra_postargs, pp_opts)
> > File "/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 513, in unix_wrap_single_compile
> > cflags = unix_cuda_flags(cflags)
> > File "/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 480, in unix_cuda_flags
> > cflags + _get_cuda_arch_flags(cflags))
> > File "/home/ace14496ms/pypy38_pp/lib/python3.8/site-packages/torch/utils/cpp_extension.py", line 1694, in _get_cuda_arch_flags
> > arch_list[-1] += '+PTX'
> > IndexError: list index out of range

did you solve this problem , I also had this problem ,could you help me?

YCC-324, 2023-04-17:
Hello, have you solved this problem?