zhulf0804/PointPillars #28: Compilation without CUDA

wojtat, 2022-12-07:
Hi!

I was wondering if there is a way to run this code with pytorch without CUDA available. I managed to make this run on a machine with CUDA, but I would like to run on my machine which doesn't have GPU capabilities. While running the `python setup.py develop` I encountered an error.
```OSError: CUDA_HOME environment variable is not set. Please set it to your CUDA install root.```
I have noticed the `setup.py` script has `CUDAExtension` in it, which I suspect might be a problem.
I am not very familiar with pytorch or CUDA, so any help would be appreciated.

Thanks in advance!

zhulf0804, 2022-12-14:
I think modifying the file `https://github.com/zhulf0804/PointPillars/blob/main/ops/setup.py` to `CppExtension` format can solve the above problem.

Please refer to PyTorch tutorial to know how to modify the code .

Best.

wojtat, 2022-12-28:
Hi,

I tried modifying the files. It seems to work for voxelization because it has a CPU implementation. I don't see a CPU iou3d implementation, so I don't think that works. Is there an implementation of iou3d that doesn't use CUDA?

Thanks!

gopalkumr, 2023-05-11:
i want to compile without cuda, i have modified code like this 

`from setuptools import setup
from torch.utils.cpp_extension import BuildExtension, CUDAExtension

setup(
    name='pointpillars',
    ext_modules=[
        CUDAExtension(
            name='voxel_op', 
            sources=['voxelization/voxelization.cpp',
                     'voxelization/voxelization_cpu.cpp',
                     'voxelization/voxelization_cuda.cu',
                    ],
            define_macros=[('WITH_CUDA', None)]    
        ),
        CUDAExtension(
            name='iou3d_op', 
            sources=['iou3d/iou3d.cpp',
                     'iou3d/iou3d_kernel.cu',
                    ],
            define_macros=[('WITH_CUDA', None)]    
        )
    ],
    cmdclass={
        'build_ext': BuildExtension
    })`

but still i'm getting this error

> (py388) PS E:\github\PointPillars\ops> python setup.py develop                                                                                
Traceback (most recent call last):
  File "setup.py", line 7, in <module>
    CUDAExtension(
  File "E:\anaconda\envs\py388\lib\site-packages\torch\utils\cpp_extension.py", line 988, in CUDAExtension
    library_dirs += library_paths(cuda=True)
  File "E:\anaconda\envs\py388\lib\site-packages\torch\utils\cpp_extension.py", line 1110, in library_paths
    paths.append(_join_cuda_home(lib_dir))
  File "E:\anaconda\envs\py388\lib\site-packages\torch\utils\cpp_extension.py", line 2130, in _join_cuda_home
    raise EnvironmentError('CUDA_HOME environment variable is not set. '
OSError: CUDA_HOME environment variable is not set. Please set it to your CUDA install root.

jonasdieker, 2023-11-14:
@wojtat could you share what you did to at least make the voxelization work? Did you end up finding/writing an alternative function for the iou3d?

punithsekar, 2025-03-26:
Any updates on this?