zhulf0804/PointPillars #29: Compile error on windows

blueeaglex, 2022-12-08:
When i tried to run the cmd "python setup.py develop", I got this:

>
(Alog3D) D:\AlogLab\PointPillars\ops>python setup.py develop
running develop
running egg_info
writing pointpillars.egg-info\PKG-INFO
writing dependency_links to pointpillars.egg-info\dependency_links.txt
writing top-level names to pointpillars.egg-info\top_level.txt
reading manifest file 'pointpillars.egg-info\SOURCES.txt'
writing manifest file 'pointpillars.egg-info\SOURCES.txt'
running build_ext
error: [WinError 2] 系统找不到指定的文件。

(Alog3D) D:\AlogLab\PointPillars\ops>cl
用于 x86 的 Microsoft (R) C/C++ 优化编译器 19.16.27048 版
版权所有(C) Microsoft Corporation。保留所有权利。

用法: cl [ 选项... ] 文件名... [ /link 链接选项... ]

As you see, the cl is avaliable. I got no idea.
Thanks to the implementation but how could I fix this problem?

szyilmaz, 2023-04-04:
If you are trying to make this code work in windows you must install "Desktop Development With C++" from Visual Studio Installer

blueeaglex, 2023-08-07:
> If you are trying to make this code work in windows you must install "Desktop Development With C++" from Visual Studio Installer

Thank you. Of course， a cpp compiler is necessary and I‘ve already installed it before the error occurred.  I think there maybe something wrong with the env and the script just DID NOT FIND the compiler.  I' ll try to figure it out.

eyabesbes, 2024-03-21:
> [blueeaglex](/blueeaglex)

Hi, 
I'm having the same issue.
I have already the Desktop Development with C++ already installed, Though, I'm still having this error.
Did you figure out how to solve it ?

blueeaglex, 2024-07-18:
Hi, @eyabesbes 
I just gave up and build it on my wsl2 ubuntu.  You can try it.