zhulf0804/PointPillars #31: The question about filter_db function.

wanghangege, 2022-12-14:
why use the function (filter_db) to filter the difficulty and min_points ?  Does it make bad influence to the performance?
`    def filter_db(self, db_infos):
        # 1. filter_by_difficulty
        for k, v in db_infos.items():
            db_infos[k] = [item for item in v if item['difficulty'] != -1]

        # 2. filter_by_min_points, dict(Car=5, Pedestrian=10, Cyclist=10)
        filter_thrs = dict(Car=5, Pedestrian=10, Cyclist=10)
        for cat in self.CLASSES:
            filter_thr = filter_thrs[cat]
            db_infos[cat] = [item for item in db_infos[cat] if item['num_points_in_gt'] >= filter_thr]
        
        return db_infos`

zhulf0804, 2022-12-14:
1. Difficulty value is calculate here
  https://github.com/zhulf0804/PointPillars/blob/b9948e73505c8d6bfa631ffdf76c7148e82c5942/pre_process_kitti.py#L16-L32
  And objects with difficulty = -1 are not cared as described in https://www.cvlibs.net/datasets/kitti/eval_object.php?obj_benchmark=3d
2. Filtering by min_points may influence the performance. We keep consistent with [filter_by_min_points](https://github.com/open-mmlab/mmdetection3d/blob/c9541b0db89498fdea5cafd05b7b17f7b625b858/mmdet3d/datasets/pipelines/dbsampler.py#L194-L214) in mmdet3d. Also the parameters are updated for a better performance in [pr 1166](https://github.com/open-mmlab/mmdetection3d/pull/1166/commits)
[image]

jonasdieker, 2023-06-22:
Hey @zhulf0804, do I understand this correctly:

The filtering here is only for the annotations database, which is used for the data augmentation when sampling extra ground truth bboxes and points. You do not filter out difficult bboxes from the actual ground truths of a particular frame when loading the data, right?