zhulf0804/PointPillars #39: Support on nuScenes and custom dataset

RnRi, 2023-03-08:
Hi, I'm wondering how to modify the dataloader to support nuScenes or custom dataset for training and Inference?

Thanks in advance.

zhulf0804, 2023-03-27:
Hello @RnRi, I think the key to support other datasets is to **write a dataset class** to return the necessary information (point cloud, bboxes, classes, etc.), such as the following dictionary in `kitti.py`.
```
data_dict = {
            'pts': pts,
            'gt_bboxes_3d': gt_bboxes_3d,
            'gt_labels': np.array(gt_labels), 
            'gt_names': annos_name,
            'difficulty': annos_info['difficulty'],
            'image_info': image_info,
            'calib_info': calib_info
        }
```
Best.

jonasdieker, 2023-06-22:
@RnRi Did you end up using NuScenes and did you train successfully? I have the data loading part working (and have verified it by visualising loaded frames), but I seem to not be able to overfit on a single training example, which must be due to ill define parameters.