zhulf0804/PointPillars #43: Training with nclasses 1 results in error

wojtat, 2023-04-01:
Hi, I'm trying to run `train.py --nclasses 1`, but it results in an `IndexError`:
```
Traceback (most recent call last):
  File "train.py", line 217, in <module>
    main(args)
  File "train.py", line 102, in main
    bbox_pred = bbox_pred[pos_idx]
IndexError: The shape of the mask [1928448] at index 0 does not match the shape of the indexed tensor [642816, 7] at index 0
```
It works fine with the default value of `nclasses`, but giving `1` gives this error. Can I ask for some help with fixing this please?

Xiong-lab, 2023-09-04:
I have the same problem with you, I fix "pos_idx = (batched_bbox>=0)&(batched)bbox_labels<args.nclasses)" in train.py to "pos_idx = (batched_bbox>=0)&(batched)bbox_labels<args.nclasses+1)"  to slove this problem.

Vito-Swift, 2024-01-25:
This is because forward() in pointpillars.py will always pass a 3-row assigner matrix to anchor_target(). A quick fix is to select only the first nclasses assigners to proceed the program.

```python
        if mode == 'train':
            anchor_target_dict = anchor_target(batched_anchors=batched_anchors,
                                               batched_gt_bboxes=batched_gt_bboxes,
                                               batched_gt_labels=batched_gt_labels,
                                               assigners=self.assigners[:self.nclasses],
                                               nclasses=self.nclasses)
```

closed 2024-04-06 by zhulf0804

Felixqzx, 2025-01-18:
> This is because forward() in pointpillars.py will always pass a 3-row assigner matrix to anchor_target(). A quick fix is to select only the first nclasses assigners to proceed the program.
> 
>         if mode == 'train':
>             anchor_target_dict = anchor_target(batched_anchors=batched_anchors,
>                                                batched_gt_bboxes=batched_gt_bboxes,
>                                                batched_gt_labels=batched_gt_labels,
>                                                assigners=self.assigners[:self.nclasses],
>                                                nclasses=self.nclasses)

not work, IndexError: The shape of the mask [1928448] at index 0 does not match the shape of the indexed tensor [2571264, 7] at index 0

Felixqzx, 2025-01-18:
> not work, IndexError: The shape of the mask [1928448] at index 0 does not match the shape of the indexed tensor [2571264, 7] at index 0

my nclasses is 4