zhulf0804/PointPillars #44: 关于eval结果与mmdet3d官方结果对比问题

hehualin-tut, 2023-04-10:
作者您这个工程很容易理解也很容易复现，但是我还是有个问题，我看了下最新的pillars的评估结果，发现mmdet3d精度是要略微高一些的，我看了下不同的地方大概在以下几点：
1.学习率设置，官方是0.001，这个工程是0.00025，我相信这是您调试出来的较好的lr。
2.在pillar_encoder里，目前官方代码好像是不再用offset去代替原始绝对坐标了。我不太清楚目前到低使用哪个效果较好。
3.数据增强方面，不管是sample_group还是db_filter您应该是在原始的基础上又进行了调整，不知道我的猜想是否正确。
4.分布式训练所带来的提升，这一点我无法肯定，因为我没有那么多显卡去验证。

gopalkumr, 2023-05-11:
Hi, @hehualin-tut I need help in setting up the environment and running the pre-trained inferences, It would be grateful if you can help me. If possible I would love to have a little conversation with you, your any contact information like, Email, or social media link would be grateful.

hehualin-tut, 2023-05-11:
> Hi, @hehualin-tut I need help in setting up the environment and running the pre-trained inferences, It would be grateful if you can help me. If possible I would love to have a little conversation with you, your any contact information like, Email, or social media link would be grateful. 

you could send your issue to hhlin@stud.tjut.edu.cn