zhulf0804/PointPillars #49: ModuleNotFoundError:  No module named 'ops.voxel_op'

farhaniwnl, 2023-06-26:
Hello,

I am having an issue when attempting to pre-process the data where I am getting a module not found error with .voxel_op as seen below, also other modules missing (iou3d, .process, etc) from the lines in the screenshot:

[image]

I would like to say since I saw this in another thread as well, my program does compile and I have visual studio installed. However, I don't have an NVIDIA GPU, and don't know if that is an issue (no CUDA runtime). Below is what it looks like when I compile the program:

[image]

zhulf0804, 2023-06-27:
Hello, @farhaniwnl.
I'm sorry. I am not familiar with compiling on Windows.
Best.

pxxxl, 2023-07-03:
Hello, @farhaniwnl.
Have you forgot to compile the program first?
```
cd ops
python setup.py develop
```

closed 2024-04-06 by zhulf0804