zhulf0804/PointPillars #50: AttributeError: 'NoneType' object has no attribute 'shape'

jyothsna-phd22, 2023-07-06:
Traceback (most recent call last):
  File "pre_process_kitti.py", line 163, in <module>
    main(args)
  File "pre_process_kitti.py", line 140, in main
    kitti_train_infos_dict = create_data_info_pkl(data_root, 'train', prefix, db=True)
  File "pre_process_kitti.py", line 58, in create_data_info_pkl
    image_shape = img.shape[:2]
AttributeError: 'NoneType' object has no attribute 'shape'

Kindly help me in this issue.

zhulf0804, 2023-07-07:
Hello @jyothsna-phd22, please check the if `img_path` exists.
https://github.com/zhulf0804/PointPillars/blob/b9948e73505c8d6bfa631ffdf76c7148e82c5942/pre_process_kitti.py#L56

couakua, 2023-10-11:
same problem

closed 2024-04-06 by zhulf0804