zhulf0804/PointPillars #52: Training and Validation Loss

markppetty, 2023-08-06:
Hello All,

This isn't really an issue with the code, more to do with me learning how to use it. Thus far, this code has been very easy to understand and has been very helpful for my learning,

However, is there a way I can get the data from the training and validation loss. I would like to plot the training and validation loss functions so I can see what's happening.

Thanks in advance

zhulf0804, 2023-08-06:
Hello, Tensorboard is supported when you run `python train.py --data_root your_path_to_kitti`. And summary file is generated as the following one.
https://github.com/zhulf0804/PointPillars/tree/main/pretrained/summary
You can use the summary file to visualize the training / validation loss.
Best.

markppetty, 2023-08-08:
Thanks for your help.

closed 2023-08-08 by zhulf0804