zhulf0804/PointPillars #64: AttributeError: 'NoneType' object has no attribute 'convert_from_pinhole_camera_parameters'

Leonnn12138, 2024-03-07:
please help

Traceback (most recent call last):
  File "test.py", line 138, in <module>
    main(args)
  File "test.py", line 87, in main
    vis_pc(pc, bboxes=lidar_bboxes, labels=labels)
  File "/mnt/PointPillars/utils/vis_o3d.py", line 97, in vis_pc
    vis_core(vis_objs)
  File "/mnt/PointPillars/utils/vis_o3d.py", line 59, in vis_core
    ctr.convert_from_pinhole_camera_parameters(param)
AttributeError: 'NoneType' object has no attribute 'convert_from_pinhole_camera_parameters'

ronkkkk, 2024-05-10:
I have the same problem and I don't know how to solve it

ABBoyangCD, 2024-05-22:
I have the same problem too!
我也有相同的问题！

farhadi76m, 2024-07-24:
> please help
> 
> Traceback (most recent call last): File "test.py", line 138, in main(args) File "test.py", line 87, in main vis_pc(pc, bboxes=lidar_bboxes, labels=labels) File "/mnt/PointPillars/utils/vis_o3d.py", line 97, in vis_pc vis_core(vis_objs) File "/mnt/PointPillars/utils/vis_o3d.py", line 59, in vis_core ctr.convert_from_pinhole_camera_parameters(param) AttributeError: 'NoneType' object has no attribute 'convert_from_pinhole_camera_parameters'

Did you solve it ?!

Windson9, 2024-07-26:
@farhadi76m did you solve it ? :D

ShuaibZyx, 2024-09-18:
Try open3d==0.16.0

quanquan-qq, 2024-11-07:
> Try open3d==0.16.0

没有用

rhnabil, 2025-01-24:
Anyone solved it?

@Leonnn12138 @zhulf0804

michelgokan, 2025-03-05:
I'm not sure if this is going to help but if you are running via a docker or apptainer, perhaps pass `--env DISPLAY=$DISPLAY` as a parameter when running exec command.

GaneshAdam, 2025-03-07:
I faced similar issue.
Commenting line "vis_pc(pc, bboxes=lidar_bboxes, labels=labels)" from onnx_infer.py solved problem for me.
vis_pc includes visualisation code which tries to create window and I was working on remote system with just ssh login (no UI access)