zhulf0804/PointPillars #65: accuracy of test is very low

Bennyliiiiii, 2024-03-08:
@zhulf0804 Hi, I trained according to your code, and the obtained weight file has a very good effect on the eval set, but the effect on the test set is very poor. Do you know the reason? thanks for your answer

zhulf0804, 2024-03-23:
Hello @Bennyliiiiii , I have not submitted test result on the official server before. 

I tested several samples from test set. Some detection results are good, while others are with few false positive detection boxes.

Could you please post your result on test set here ?

PoPdark, 2025-08-11:
> [@zhulf0804](https://github.com/zhulf0804) Hi, I trained according to your code, and the obtained weight file has a very good effect on the eval set, but the effect on the test set is very poor. Do you know the reason? thanks for your answer

have you solved the problem?