zhulf0804/PointPillars #68: RuntimeError: Sizes of tensors must match except in dimension 2. Got 316 and 313 (The offending index is 0

kihoon71, 2024-03-28:
Hi. I'm trying to use custom dataset with your model.
I've changed few codes where I think I have to, like
point_cloud_range, classes, etc.

When I run the command
-------------------------------------------------
python train.py --data_root own_data
-------------------------------------------------

I get error message below.
------------------------------------------------------------------------------------------------------------------------
RuntimeError: Sizes of tensors must match except in dimension 2. Got 316 and 313 (The offending index is 0)
------------------------------------------------------------------------------------------------------------------------

can I get any help?

zhulf0804, 2024-04-01:
Hello, @kihoon71, which line causes the error ?

kihoon71, 2024-04-02:
Instead of my team mate, i did upload this issue. He must know better about this issue than me, so i will mention him here.

@fb059

fb059, 2024-04-03:
Hi, I'm the one who asked for it. 

The error occures in line 181 of pointpillars.py.

I'll put total error message below.

Traceback (most recent call last):
  File "train.py", line 222, in <module>
    main(args)
  File "train.py", line 89, in main
    pointpillars(batched_pts=batched_pts, 
  File "/opt/conda/lib/python3.8/site-packages/torch/nn/modules/module.py", line 889, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/workspace/PointPillars/model/pointpillars.py", line 393, in forward
    x = self.neck(xs)
  File "/opt/conda/lib/python3.8/site-packages/torch/nn/modules/module.py", line 889, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/workspace/PointPillars/model/pointpillars.py", line 181, in forward
    out = torch.cat(outs, dim=1)
RuntimeError: Sizes of tensors must match except in dimension 2. Got 316 and 313 (The offending index is 0)

zhulf0804, 2024-04-06:
Hi, @fb059 ,  could you help to provide your config for `voxel_size` and `point_cloud_range` ?
And then `print(xi.size())` in https://github.com/zhulf0804/PointPillars/blob/57af3f50fd3442e6e5c22467e16d28033c4cc51a/model/pointpillars.py#L179

fb059, 2024-04-08:
voxel_size is 
[0.25, 0.25, 8]

and point_cloud_range is 
[-50, -50, -5, 50, 50, 3].

And when I print(xi.size()) in pointpillars.py, the output is
torch.Size([6, 128, 200, 200])

zhulf0804, 2024-04-08:
> voxel_size is [0.25, 0.25, 8]
> 
> and point_cloud_range is [-50, -50, -5, 50, 50, 3].
> 
> And when I print(xi.size()) in pointpillars.py, the output is torch.Size([6, 128, 200, 200])

Hi, so how 316 and 313 are generated, when your x_i.size() is `[6, 128, 200, 200]`?

fb059, 2024-04-09:
Sorry, my mistake.
I changed configs in pointpillars.py to

                 nclasses=2, 
                voxel_size=[0.16, 0.16, 4],
                point_cloud_range=[0, -39.68, -3, 69.12, 39.68, 1],
                max_num_points=32,
                max_voxels=(16000, 40000)):
                
and after that change

RuntimeError: Sizes of tensors must match except in dimension 2. Got 316 and 313 (The offending index is 0)

no longer occurs.
But now I get different error described below.

--------------------------------------------------------------
Traceback (most recent call last):
  File "train.py", line 215, in <module>
    main(args)
  File "train.py", line 154, in main
    pointpillars(batched_pts=batched_pts, 
  File "/opt/conda/lib/python3.8/site-packages/torch/nn/modules/module.py", line 889, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/workspace/PointPillars/model/pointpillars.py", line 396, in forward
    pillar_features = self.pillar_encoder(pillars, coors_batch, npoints_per_pillar)
  File "/opt/conda/lib/python3.8/site-packages/torch/nn/modules/module.py", line 889, in _call_impl
    result = self.forward(*input, **kwargs)
  File "/workspace/PointPillars/model/pointpillars.py", line 92, in forward
    pooling_features = torch.max(features, dim=-1)[0] # (p1 + p2 + ... + pb, out_channels)
RuntimeError: cannot perform reduction function max on tensor with no elements because the operation does not have an identity

zhulf0804, 2024-04-10:
Hi @fb059, it seems there are no elements in tensor `features`. 
Please help check features.size() in 
https://github.com/zhulf0804/PointPillars/blob/57af3f50fd3442e6e5c22467e16d28033c4cc51a/model/pointpillars.py#L76

and 

https://github.com/zhulf0804/PointPillars/blob/57af3f50fd3442e6e5c22467e16d28033c4cc51a/model/pointpillars.py#L91

fb059, 2024-04-11:
Hi, @zhulf0804 
First, I misunderstood my point_cloud_range(min x,y,z values and max x,y,z values), so I fixed it right. 
```py
point_cloud_range=[-39.9442, -39.7883, -2.3986, 39.8011, 39.8065, 6.9456],
```

And as you recommanded, I checked features.size()
The first features.size() looks like 

torch.Size([583, 64, 9])
torch.Size([892, 64, 9])
torch.Size([879, 64, 9])
torch.Size([905, 64, 9])
torch.Size([626, 64, 9])
torch.Size([789, 64, 9])
torch.Size([742, 64, 9])
torch.Size([697, 64, 9])
torch.Size([632, 64, 9])
torch.Size([636, 64, 9])
torch.Size([623, 64, 9])
torch.Size([644, 64, 9])
torch.Size([679, 64, 9])
torch.Size([916, 64, 9])
torch.Size([871, 64, 9])
torch.Size([720, 64, 9])
torch.Size([704, 64, 9])
torch.Size([896, 64, 9])
torch.Size([833, 64, 9])
torch.Size([855, 64, 9])
torch.Size([880, 64, 9])
torch.Size([762, 64, 9])
torch.Size([872, 64, 9])
torch.Size([642, 64, 9])
torch.Size([825, 64, 9])
torch.Size([634, 64, 9])
torch.Size([751, 64, 9])
torch.Size([608, 64, 9])
torch.Size([474, 64, 9])
torch.Size([663, 64, 9])
torch.Size([540, 64, 9])
torch.Size([982, 64, 9])
torch.Size([772, 64, 9])
torch.Size([813, 64, 9])
torch.Size([0, 64, 9])
Traceback (most recent call last):
-------------------------------

And the second features.size() looks like

torch.Size([583, 64, 9])
torch.Size([583, 64, 64])
torch.Size([892, 64, 9])
torch.Size([892, 64, 64])
torch.Size([879, 64, 9])
torch.Size([879, 64, 64])
torch.Size([905, 64, 9])
torch.Size([905, 64, 64])
torch.Size([626, 64, 9])
torch.Size([626, 64, 64])
torch.Size([789, 64, 9])
torch.Size([789, 64, 64])
torch.Size([742, 64, 9])
torch.Size([742, 64, 64])
torch.Size([697, 64, 9])
torch.Size([697, 64, 64])
torch.Size([632, 64, 9])
torch.Size([632, 64, 64])
torch.Size([636, 64, 9])
torch.Size([636, 64, 64])
torch.Size([623, 64, 9])
torch.Size([623, 64, 64])
torch.Size([644, 64, 9])
torch.Size([644, 64, 64])
torch.Size([679, 64, 9])
torch.Size([679, 64, 64])
torch.Size([916, 64, 9])
torch.Size([916, 64, 64])
torch.Size([871, 64, 9])
torch.Size([871, 64, 64])
torch.Size([720, 64, 9])
torch.Size([720, 64, 64])
torch.Size([704, 64, 9])
torch.Size([704, 64, 64])
torch.Size([896, 64, 9])
torch.Size([896, 64, 64])
torch.Size([833, 64, 9])
torch.Size([833, 64, 64])
torch.Size([787, 64, 9])
torch.Size([787, 64, 64])
torch.Size([855, 64, 9])
torch.Size([855, 64, 64])
torch.Size([880, 64, 9])
torch.Size([880, 64, 64])
torch.Size([762, 64, 9])
torch.Size([762, 64, 64])
torch.Size([872, 64, 9])
torch.Size([872, 64, 64])
torch.Size([642, 64, 9])
torch.Size([642, 64, 64])
torch.Size([825, 64, 9])
torch.Size([825, 64, 64])
torch.Size([634, 64, 9])
torch.Size([634, 64, 64])
torch.Size([751, 64, 9])
torch.Size([751, 64, 64])
torch.Size([608, 64, 9])
torch.Size([608, 64, 64])
torch.Size([474, 64, 9])
torch.Size([474, 64, 64])
torch.Size([663, 64, 9])
torch.Size([663, 64, 64])
torch.Size([935, 64, 9])
torch.Size([935, 64, 64])
torch.Size([540, 64, 9])
torch.Size([540, 64, 64])
torch.Size([982, 64, 9])
torch.Size([982, 64, 64])
torch.Size([772, 64, 9])
torch.Size([813, 64, 9])
torch.Size([813, 64, 64])
torch.Size([0, 64, 9])
torch.Size([0, 64, 64])
Traceback (most recent call last):
---------------------------------------------

zhulf0804, 2024-04-14:
Hi @fb059, as the final feature'size is [0, 64, 9], it seems that the data is abnormal. Please help check it. Thanks.