zhulf0804/PointPillars #74: 训练角度有问题，代码有bug

superchenyan, 2024-07-23:
bbox_pred[:, -1] = torch.sin(bbox_pred[:, -1].clone()) * torch.cos(batched_bbox_reg[:, -1].clone())
batched_bbox_reg[:, -1] = torch.cos(bbox_pred[:, -1].clone()) * torch.sin(batched_bbox_reg[:, -1].clone())
bbox_dir_cls_pred = bbox_dir_cls_pred[pos_idx]

superchenyan, 2024-07-23:
batched_bbox_reg[:, -1] = torch.cos(bbox_pred[:, -1].clone()) * torch.sin(batched_bbox_reg[:, -1].clone())这行的bbox_pred[:, -1].clone()已经被上面一行修改了

zhulf0804, 2024-07-24:
> batched_bbox_reg[:, -1] = torch.cos(bbox_pred[:, -1].clone()) * torch.sin(batched_bbox_reg[:, -1].clone())这行的bbox_pred[:, -1].clone()已经被上面一行修改了

你好，@superchenyan，
是的，这里的确存在问题，和https://github.com/zhulf0804/PointPillars/issues/38 这个问题类似。
欢迎提PR修正和补充实验结果。

祝好！