zhulf0804/PointPillars #76: Bug about box_collision_test

SubChange, 2024-07-31:
When using funtions to detect bbox collision, I found:

```python
from utils.process import box_collision_test

bbox = np.array([[[205.73,   6.56],
                        [205.73,   9.36],
                        [214.53,   9.36],
                        [214.53,   6.56]],
                    [[205.75,   6.66],
                        [205.75,   9.36],
                        [214.35,   9.36],
                        [214.35,   6.66]]], dtype=np.float32)
mtx = box_collision_test(bbox, bbox)
print(mtx)
```
**Result**
```text
[[False, False],
[False, False]]
```
I know box[1] overlap with box[2], but the result is False. Can anyone help to solve this problem, Thank you very much.

SubChange, 2024-08-01:
I rewrite this function based on SAT thoery,
```python

def box2d_sat_collision_test(box1, box2):
    # box1: 4 x 2
    boxes = np.vstack((box1, box2)) # 8*2

    rot90 = np.array([[0.,1.],[-1.,0.]], dtype=np.float32)
    line_vec = np.array([box1[1]-box1[0],
                     box1[2]-box1[1],
                     box2[1]-box2[0],
                     box2[2]-box2[1]]) # n x 
    nl = np.sqrt(np.sum(line_vec**2, axis=1))
    line_vec = line_vec / nl[:,None]
    nvec = line_vec @ rot90.T # 4x2

    projL = np.dot(boxes, nvec.T) # 

    x1_min, x1_max = np.min(projL[:4, :], axis=0), np.max(projL[:4, :], axis=0)
    x2_min, x2_max = np.min(projL[-4:, :], axis=0), np.max(projL[-4:, :], axis=0)

    if np.any(x1_min>x2_max) or np.any(x1_max<x2_min):
        return False

    return True
    

# modified from https://github.com/open-mmlab/mmdetection3d/blob/master/mmdet3d/datasets/pipelines/data_augment_utils.py#L31
# @numba.jit(nopython=True)
def box_collision_test(boxes, qboxes, clockwise=True, ignore_diag=True):
    """Box collision test.
    Args:
        boxes (np.ndarray): Corners of current boxes. # (n1, 4, 2)
        qboxes (np.ndarray): Boxes to be avoid colliding. # (n2, 4, 2)
        clockwise (bool, optional): Whether the corners are in
            clockwise order. Default: True.
    return: shape=(n1, n2)
    """
    N = boxes.shape[0]
    K = qboxes.shape[0]
    ret = np.zeros((N, K), dtype=np.bool_)
    # slices = np.array([1, 2, 3, 0])
    # lines_boxes = np.stack((boxes, boxes[:, slices, :]),
    #                        axis=2)  # [N, 4, 2(line), 2(xy)] # 新增了一个维度
    # lines_qboxes = np.stack((qboxes, qboxes[:, slices, :]), axis=2)
    # # vec = np.zeros((2,), dtype=boxes.dtype)
    # boxes_standup = bevcorner2alignedbbox(boxes)
    # qboxes_standup = bevcorner2alignedbbox(qboxes)
    for i in range(N):
        for j in range(K):
            if ignore_diag and i==j:
                continue
            ret[i, j] = box2d_sat_collision_test(boxes[i], qboxes[j])
            # # calculate standup first
            # # aabb 碰撞
            # iw = (
            #     min(boxes_standup[i, 2], qboxes_standup[j, 2]) -
            #     max(boxes_standup[i, 0], qboxes_standup[j, 0]))
            # if iw > 0:
            #     ih = (
            #         min(boxes_standup[i, 3], qboxes_standup[j, 3]) -
            #         max(boxes_standup[i, 1], qboxes_standup[j, 1]))
            #     if ih > 0:
            #         for k in range(4):
            #             for box_l in range(4):
            #                 A = lines_boxes[i, k, 0]
            #                 B = lines_boxes[i, k, 1]
            #                 C = lines_qboxes[j, box_l, 0]
            #                 D = lines_qboxes[j, box_l, 1]
            #                 acd = (D[1] - A[1]) * (C[0] -
[... 43 lines cut ...]
            #                     cross_ = []
            #                     for k in range(4):  # corner k in qboxes
            #                         vec = qboxes[j, k] - qboxes[j, (k + 1) % 4]
            #                         if clockwise:
            #                             vec = -vec
            #                         cross = vec[1] * (
            #                             qboxes[j, k, 0] - boxes[i, box_l, 0])
            #                         cross -= vec[0] * (
            #                             qboxes[j, k, 1] - boxes[i, box_l, 1])
            #                         # if cross < 0:  #
            #                         #     qbox_overlap_box = False
            #                         #     break
            #                         cross_.append(cross)
            #                     # if qbox_overlap_box is False:
            #                     #     break
            #                     if np.any(np.array(cross_)>0):
            #                         box_overlap_qbox = False
            #                         break
            #                 if qbox_overlap_box:
            #                     ret[i, j] = True  # collision.
            #             else:
            #                 ret[i, j] = True  # collision.
    return ret

```

zhulf0804, 2024-08-02:
Hi @SubChange, nice implementation based on SAT thoery. 

There might be some issues with the handling of collinear cases in the origin code, and I will fix it soon.

Best.