zhulf0804/PointPillars #77: Point Cloud visualization problem

MiniDudi, 2024-08-20:
i'm trying to convert the point cloud results into a kitti dataset .bin file and for some reason it's returning me a error when running the test:

`TypeError: tuple indices must be integers or slices, not str` #21 

convertion code:
```
class MinimalSubscriber(Node):

    def __init__(self):
        super().__init__('minimal_subscriber')
        self.subscription = self.create_subscription(
            PointCloud2,
            'velodyne_points',
            self.listener_callback,
            10)

    def convert_pointcloud(self, msg):

        pc = pc2.read_points(msg, field_names=("x", "y", "z", "intensity"), skip_nans=True)
        pc = np.array(list(pc), dtype=np.float32)

        x = pc[:, 0]
        y = pc[:, 1]
        z = pc[:, 2]
        intensity = pc[:, 3]
        
        arr = np.zeros(x.shape[0] * 4, dtype=np.float32)
        arr[::4] = x
        arr[1::4] = y
        arr[2::4] = z
        arr[3::4] = intensity
        
        bin_file_name = 'point_cloud_data.bin'
        with open(bin_file_name, 'wb') as f:
            arr.tofile(f)

    def listener_callback(self, msg):

        self.convert_pointcloud(msg)
```

To resolve the error above, i removed the following line in test.py:

`#model.eval()` #48 

After this, the inference worked and displayed the window with the detected objects. But, all the point cloud is showing in white.

[image]

### What am i doing wrong? Do you guys have any idea?
Thank you in advance.

kauanBestel, 2024-08-20:
I'm facing the same problem.

vutrungtam619, 2025-09-03:
Sorry, do you know the solution now?