zhulf0804/PointPillars #97: Hello, I have this Error... While Running this code'python pre_process_kitti.py --data_root kitti'  . please help me...

duchuyy05, 2025-05-01:
(pp_env) D:\PointPillars>python pre_process_kitti.py --data_root D:\PointPillars\pointpillars\kitti
Traceback (most recent call last):
  File "pre_process_kitti.py", line 11, in <module>
    from pointpillars.utils import read_points, write_points, read_calib, read_label, \
  File "D:\PointPillars\pointpillars\utils\__init__.py", line 3, in <module>
    from .process import bbox_camera2lidar, bbox3d2bevcorners, box_collision_test, \
  File "D:\PointPillars\pointpillars\utils\process.py", line 7, in <module>
    from pointpillars.ops.iou3d_module import boxes_overlap_bev, boxes_iou_bev
  File "D:\PointPillars\pointpillars\ops\__init__.py", line 1, in <module>
    from .voxel_module import Voxelization
  File "D:\PointPillars\pointpillars\ops\voxel_module.py", line 5, in <module>
    from pointpillars.ops.voxel_op import hard_voxelize
ModuleNotFoundError: No module named 'pointpillars.ops.voxel_op'

duchuyy05, 2025-05-01:
Hello, I have this Error... While Running this code'python pre_process_kitti.py --data_root kitti' please help me...

sihogu, 2025-05-14:
```
python setup.py build_ext --inplace
pip install .
```